ANYbotics/kindr #60: Bug in Kindr Cheat Sheet

nikolicj, 2015-08-27:
There seems to be a bug in the Cheat Sheet pdf, on page two: http://ethz-asl-lr.bitbucket.org/kindr/cheatsheet_latest.pdf , second last line. It should be:

`C_{AB} = (2 p_0 ^2 - 1) I + ...`

HannesSommer, 2015-08-27:
Great catch, @nikolicj !
I agree with that there is a bug. But I come to a different conclusion. Based on the rotation expression for the rotation of ```v``` with ```p = p * (0, v) * p^{-1} = Q(q) \bar Q(q^{-1}) (0, v)``` it follows that we are looking for the lower right 3x3 sub matrix of matrix ```Q(q) \bar Q(q^{-1})```. This yields, in my opinion, with the arrowed fat p written as p:

```C = p_0^2 I +  2p_0 \hat p + (\hat p)^2 + p p^T```

The good news: this seems to fit the last line of page two (bu I only checked some entries: 1,1; 2,2; 3,3;1,2). 
Ah and we have the analogous problem in the first line of page three...

nikolicj, 2015-08-28:
OK!

My suggestion was based (wrongly) on (78) in http://www-users.cs.umn.edu/~trawny/Publications/Quaternions_3D.pdf .

nikolicj, 2015-08-28:
Check.

It seems equivalent to `(2 p_0^2 - 1) I + 2 p_0 hat p + 2 p p^T`.

HannesSommer, 2015-08-28:
Thanks @nikolicj . Great! You're right. I didn't see it but it is the same, basically because of : 

```a x (b x c) = (a^T c) b - (a^T b) c```
This implies for arbitrary ```c```:
```\hat p^2 * c= p x ( p x  c) = (p^T c)p - p^2 c = p(p^T c) - p^2 c = p p^T c + (q_0^2 - 1) c = (p p^T c + (q_0^2 - 1) I) c```

(the second last equality requires ```1 = q_0^2 + q^2```, which is the unit length property of ```p```)

So we get, as c was arbitrary:
```
\hat p^2 = pp^T + (q_0^2 - 1) I
```

This is exactly the difference of the two versions up there.

Great stuff! 

I'd suggest we put it both in there, because it isn't that obvious and one or the other might be more suitable for some application.

HannesSommer, 2016-06-24:
Oh my, the very same bug seems to be in quite prominent position https://en.wikipedia.org/wiki/Euler%E2%80%93Rodrigues_formula.

gehrinch, 2016-06-24:
This should be closed, right?

HannesSommer, 2016-06-24:
Wiki: Ah, I was just again confusing `pp^T` with `(\hat p)^2`. 
Yes that seems solved in the current tex.

closed 2016-06-24 by gehrinch