ANYbotics/kindr #66: rovio does not work

cirpote, 2016-06-30:
The problem is that the last commits on the kindr repository make many header errors compiling the rovio repository.  I tryied both "sudo apt-get install ros-indigo-kindr-*" and by cmake and catkin.

gehrinch, 2016-06-30:
There was an interface change from kindr 0.0.1 to kindr 1.0.0! Please create an issue in the rovio lib. Do you also complain to Microsoft or Apple if you cannot install a thirdparty software on your system? Please checkout the release 0.0.1 of kindr to build rovio. The good news is that rovio is being adapted to the new kindr version. It should be released soonish.

closed 2016-06-30 by gehrinch

bloesch, 2016-07-06:
Rovio has been updated and should now compile with kindr1.