ANYbotics/kindr #91: EIGEN3_INCLUDE_DIR

samuelba, 2017-08-10:
@schneith You recently changed Eigen3 to Eigen for whatever reason.

1. Why did you change this? `find_package(Eigen3 REQUIRED)` works perfectly fine and there isn't really a need for an extra FindEigen.cmake
2. If we stick to your FindEigen.cmake at least kindr_include_dirs must be adapted https://github.com/ethz-asl/kindr/blob/master/CMakeLists.txt#L117

schneith, 2017-08-10:
1)
The reason for this change was that the provided Eigen script here
https://github.com/ethz-asl/kindr/blob/master/cmake/FindEigen.cmake
was not in the search path of cmake. This means that the the following line 
https://github.com/ethz-asl/kindr/blob/master/CMakeLists.txt#L34
would not be able to find the FindEigen.cmake script and would per fall back to look whether in can find anything in the default system paths. On a clean Ubuntu (at least in 14.04), you will not find such a script for Eigen and the compilation of kindr will therefore fail.

2)
Yes, you are right. I missed that line that should also be changed to:
```
set(kindr_include_dirs ${CMAKE_INSTALL_PREFIX}/include ${EIGEN_INCLUDE_DIR})
```
Sorry for that bug.

schneith, 2017-08-10:
You can find the fix here: https://github.com/ethz-asl/kindr/pull/92

samuelba, 2017-08-10:
ok

closed 2017-08-10 by samuelba