ANYbotics/kindr #111: kindr building

thq1995, 2018-10-13:
Having an issue with building kindr. I cannot use catkin_make to compile a package for elevation_mapping. Terminal shows that kindr and other packages are non-homogenous

LuisSierraP, 2019-03-01:
I have the same issue, Have you been able to solve it?

samuelba, 2019-03-04:
We use [catkin tools](https://catkin-tools.readthedocs.io/en/latest/) and have no problems. Can you provide de terminal output?

LuisSierraP, 2019-03-05:
I'm using catkin_make to build the workspace and the output I get is:
`-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - kindr (plain cmake)
-- ~~  - kindr_msgs
-- ~~  - kindr_ros
-- ~~  - ground_removal
-- ~~  - kindr_rviz_plugins
-- ~~  - multi_dof_joint_trajectory_rviz_plugins
-- ~~  - velocity_costmap
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/luis/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/luis/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
`

samuelba, 2019-03-05:
Seems to be a catkin_make problem. I guess the only solution is to either use catkin_make_isolated or use the catkin tools as we do.