ANYbotics/kindr #116: Doxygen Latex missing

jolau, 2019-07-08:
On https://docs.leggedrobotics.com/kindr/index.html the images of the latex formulas are missing and therefore only shows the (incomprehensible) latex code. Could you please fix it? Otherwise the docs are not really usable.

remod, 2019-07-09:
Thanks for reporting this. @pleemann Is `texlive-latex-base` installed on the build server? It is required to compile the latex formulas in doxygen.

pleemann, 2019-07-09:
No its not, but adding it as <code_depend> should do the trick

remod, 2019-07-17:
Do you mean <doc_depend>?

pleemann, 2019-07-17:
Yes, sorry

remod, 2019-07-30:
It does not seem to have solved the issue, any ideas?

jolau, 2019-08-08:
Maybe you could also try to use MathJax instead, by enable USE_MATHJAX: http://www.doxygen.nl/manual/config.html#cfg_use_mathjax

pleemann, 2019-08-08:
It should work, but the docs on https://docs.leggedrobotics.com/kindr/index.html are not automatically updated anymore. The documentation with proper latex formulas will be available under new URL after next release (latest end of this month). Stay tuned and sorry for any inconvenience.

hogabrie, 2019-09-23:
I updated the README with the new URL.  See 0c16c8232d5404b09284cb6a441365172e163b2b

closed 2019-09-23 by hogabrie