ANYbotics/kindr #119: Operations for one-dimensional Time

v-lopez, 2020-01-29:
First of all thanks for this great work, it's very helpful.

If I understood correctly, the way to get the proper way of getting the translation from a velocity and a time duration is the following:

```
  kindr::Time3D delta;
  kindr::Velocity3D vel;
  kindr::Position3D pos = vel.elementwiseMultiplication(delta);
```

Using a 3D vector for time for this operations is usually unnecessary and can obfuscate a bit the code.

I have some local modifications that allow the following:

```
  kindr::TimeDelta delta;
  kindr::Velocity3D vel;
  kindr::Position3D pos = vel*delta; // GOOD
  kindr::Position3D pos2 = vel*1.0; // does not compile because 1.0 is not TimeDelta
  kindr::Velocity3D pos3 = vel*delta; // does not compile because return type is not converitble to Velocity3D
```

Am I missing something? If not, would there be interest in a MR with this feature?

remod, 2020-06-04:
Does your code work with the recent changes introduced in https://github.com/ANYbotics/kindr/commit/fcea3c9bb9d7af93c6a781288a2ec0bf334c0d92?

remod, 2020-06-04:
You can now use 1D vectors like scalars.

The following snippet compiles:
```
kindr::TimeD delta(0.5);
kindr::Velocity3D vel(1.0, 2.0, 3.0);
kindr::Position3D pos1 = vel * delta;
kindr::Position3D pos2 = delta * vel;
std::cout << pos1 << std::endl;
std::cout << pos2 << std::endl;
```

v-lopez, 2020-06-04:
That's great, that's exactly what I needed. 
I managed to get just for Time type without modifying kindr itself, but this implementation is better in every way.

closed 2020-06-04 by v-lopez